CNflysky/redroid-rk3588 #1: dmesg 刷屏

txlos, 2024-03-07:
orangepi 5 plus, 用的官方的ubuntu 22.04,  mali驱动，xserver桌面
貌似不影响使用，就是一直刷屏。

dmesg：
[16520.628515] logd: logdr: UID=10052 GID=10052 PID=1062 n tail=0 logMask=4 pid=0 start=0ns deadline=0ns
[16522.618411] ------------[ cut here ]------------
[16522.618430] WARNING: CPU: 7 PID: 2901 at drivers/gpu/arm/bifrost/mali_kbase_mem_migrate.c:606 kbase_mem_migrate_set_address_space_ops+0x8c/0x94
[16522.618432] Modules linked in: xt_nat veth xt_conntrack nf_conntrack_netlink xt_addrtype br_netfilter rfcomm xt_CHECKSUM nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_counter nft_compat bridge stp llc nf_tables nfnetlink bnep overlay zstd binfmt_misc squashfs btusb btrtl btbcm btintel bluetooth pwm_fan iwlmvm iwlwifi sch_fq_codel fuse ip_tables ipv6 pgdrv r8125
[16522.618489] CPU: 7 PID: 2901 Comm: Chrome_InProcGp Tainted: G        W         5.10.160-rockchip-rk3588 #1.0.8
[16522.618491] Hardware name: RK3588 OPi 5 Plus (DT)
[16522.618495] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO BTYPE=--)
[16522.618498] pc : kbase_mem_migrate_set_address_space_ops+0x8c/0x94
[16522.618500] lr : kbase_mem_migrate_set_address_space_ops+0x34/0x94
[16522.618502] sp : ffffffc01b693a70
[16522.618504] x29: ffffffc01b693a70 x28: ffffff83be304c40 
[16522.618507] x27: 0000000000000000 x26: ffffff81e2844510 
[16522.618509] x25: 0000000000000006 x24: ffffff8175b238d8 
[16522.618513] x23: 0000000000000078 x22: ffffffc0091b0ee0 
[16522.618515] x21: ffffff81083e30b0 x20: ffffff8308dde000 
[16522.618518] x19: ffffff81083e8000 x18: 0000000000000000 
[16522.618521] x17: 0000000000000000 x16: 0000000000000000

CNflysky, 2024-03-07:
我这里也有，猜测是主机和容器共用了GPU某个部分导致冲突了，但不影响使用，就没管了

same situation here. i guess the container and host uses shared gpu stuff. anyway, it still works though.

CNflysky, 2024-04-30:
stale

closed 2024-04-30 by CNflysky