CNflysky/redroid-rk3588 #2: 虚拟wifi生效但app无网络 Virtual wifi works but no network for apps

RongBeiJun, 2024-03-13:
我尝试按照readme启用虚拟wifi成功，但是任何应用都没有网络连接
I tried to follow the readme to enable virtual wifi successfully, but there is no network connection for any apps
设置中显示virtwifi网络连接受限，而在一些终端app（或者docker exec -it  redroid sh）执行如ping 39.156.66.14时可以看到返回结果
The setting shows that the virtwifi network connection is restricted, while in some terminal apps (or docker exec -it redroid sh) you can see the return result when executing e.g. ping 39.156.66.14
请问您有什么建议或解决方案吗
Do you have any suggestions or solutions?

[image]
[image]
[image]
[image]
[image]
[image]

CNflysky, 2024-03-13:
你用的哪个镜像？13的话我已经把captive portal server地址改成小米的了，如果你用的是12的话:  
```bash
adb shell settings put global captive_portal_https_url https://connect.rom.miui.com/generate_204
adb shell settings put global captive_portal_http_url https://connect.rom.miui.com/generate_204
```
这样就可以解决无互联网连接的问题了
顺便，虚拟WiFi的DNS服务器指定的是`8.8.8.8`，检查一下你那边能不能正常访问。

which image you're using now? if `redroid 13` then i've already changed captive portal server to xiaomi's. if you are using 12, you can switch to xiaomi's captive portal server:
```bash
adb shell settings put global captive_portal_https_url https://connect.rom.miui.com/generate_204
adb shell settings put global captive_portal_http_url https://connect.rom.miui.com/generate_204
```
then the "no internet" prompt will disappear.  
BTW, DNS server address in container is `8.8.8.8`, also check you can access that address.

RongBeiJun, 2024-03-13:
感谢回复，这或许不是captive portal server的原因
Thanks for the reply, this may not be the reason for captive portal server
我在提issue后从orangepi的ubuntu更换到了`Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal`
I changed from orangepi's ubuntu to `Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal` after issue
并且使用了cnflysky/redroid-rk3588:13.0.0-latest，在查看logcat后发现了一些关于iptables的错误
And using cnflysky/redroid-rk3588:13.0.0-latest, after checking logcat I found some error about iptables
我尝试修复了一部分（通过modprop加载模块解决了如`can't initialize iptables table 'nat'`）但仍有部分未能解决
I tried to fix some of them (loading modules via modprop solved e.g. `can't initialize iptables table 'nat'`) but still some of them were not resolved
（关于DNS 8.8.8.8，这里可以访问）
(For DNS 8.8.8.8, access it here)
我对于虚拟wifi和iptables之间的工作方式不了解，我应该做些什么？
I have no idea about how virtual wifi and iptables work with each other, what should I do?
[image]
[image]
[image]

CNflysky, 2024-03-13:
iptablesrestore报错是正常的，不用管
你关掉虚拟WiFi之后能上网吗？主机上切换到iptables-legacy试试？

iptablesrestore error is normal
can you connect to internet after disable virtual wifi? also try switch to `iptables-legacy` on your host.

CNflysky, 2024-03-13:
能提供一下你的内核deb包吗 我尝试在这边复现一下

can you provide your kernel deb package? im trying to reproduce your issue.

RongBeiJun, 2024-03-13:
当然可以，链接https://www.123pan.com/s/G0EDVv-qWHKd.html
Sure, link
因为超过25MB，限制上传
Limited to uploading because it's over 25MB
如果我的内核编译是错误的，希望您可以提供内核deb，或者告知我正确的做法
If my kernel compilation is wrong, I would like you to provide the kernel deb, or inform me of the correct way to do it, thanks

CNflysky, 2024-03-13:
无法复现。操作步骤如下：
- 安装`Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal` 系统
- 安装你提供的内核包
- 重启
- 更换国内源(清华源)
- 安装docker `sudo apt install docker.io`
- 拉取镜像 `docker pull cnflysky/redroid-rk3588:13.0.0-latest`
- `docker run -d -p 5555:5555 -v ~/redroid-data:/data --name redroid --privileged cnflysky/redroid-rk3588:13.0.0-latest androidboot.redroid_height=1920 androidboot.redroid_width=1080 androidboot.redroid_virt_wifi=1`
- 复制Mali GPU固件 `docker cp redroid:/vendor/etc/firmware/mali_csffw.bin /lib/firmware`
- 等待系统启动完成
- 使用scrcpy，可以正常连接网络。

can't reproduce. steps:
- install `Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal` 
- install kernel deb package you have provided
- reboot
- switch to Chinese debian sources
- install docker `sudo apt install docker.io`
- `docker pull cnflysky/redroid-rk3588:13.0.0-latest`
- `docker run -d -p 5555:5555 -v ~/redroid-data:/data --name redroid --privileged cnflysky/redroid-rk3588:13.0.0-latest androidboot.redroid_height=1920 androidboot.redroid_width=1080 androidboot.redroid_virt_wifi=1`
- copy Mali GPU firmware `docker cp redroid:/vendor/etc/firmware/mali_csffw.bin /lib/firmware`
- wait until Android boot up
- use scrcpy to stream screen, network is ok

[image]

RongBeiJun, 2024-03-13:
您好，在您尝试复现问题的同时，我也在做其他的尝试
While you are trying to reproduce the problem, I am making other attempts
结果是app已经成功连接网络
The result is that the app has successfully connected to the network
成功的尝试的环境如下
The environment for a successful attempt is as follows
系统为`Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal`
system is `Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal`
开启mac80211_hwsim的内核
Enabling the kernel for mac80211_hwsim
docker run部分依然遵从readme（但使用cnflysky/redroid-rk3588:13.0.0-latest）
The docker run part still follows the readme (but uses cnflysky/redroid-rk3588:13.0.0-latest)
**保证只有无线连接**（猜测此项为问题主要原因，此前有线连接与无线连接同时存在并且连接在相同局域网）
**Ensure that there is only a wireless connection** (guessing that this is the main cause of the problem, as both wired and wireless connections existed and were connected to the same LAN earlier)
redroid中app网络可用
network availability

17603127956, 2024-03-14:
> 您好，在您尝试复现问题的同时，我也在做其他的尝试 While you are trying to reproduce the problem, I am making other attempts 结果是app已经成功连接网络 The result is that the app has successfully connected to the network 成功的尝试的环境如下 The [...]

请问如何保证只有无线连接，我也遇到了相同的问题。
May I ask how to ensure only wireless connection? I have also encountered the same problem

CNflysky, 2024-03-14:
将主机切换到`iptables-legacy`随后重启 或 使用`sudo modprobe iptable_nat`加载模块，之后再创建容器。已经加到README里面了。  

switch to `iptables-legacy` in your host os or `sudo modprobe iptable_nat` load nat module, then create container. added to README.

CNflysky, 2024-06-23:
理论上，可以通过在framework层对connectionmanager类进行改动，使其对app返回一个虚假的WiFi连接以达到虚拟WiFi的效果。但是由于本人不擅长Android开发，只能搁置。如果有哪位大佬愿意帮助，欢迎PR。

It's possible to implement `Virtual WiFi` through this way: return a fake WiFi connection by patching into `ConnectionManager` class in `AOSP Framework`. but I don't know how to do it. PR is welcomed.

CNflysky, 2024-07-07:
Completed.
[commit](https://github.com/CNflysky/redroid-rk3588/commit/a3d32c08923eb65edac64f53549de947361a94f6)

closed 2024-07-07 by CNflysky