CNflysky/redroid-rk3588 #3: 碧蓝航线闪退

CoronaAustralis, 2024-04-29:
**Describe the bug**
**描述Bug**
碧蓝航线闪退，其他游戏都没有问题

**Required Information**
**基本信息**
Android13

- OS Information(`Debian`、`Ubuntu`……)
rk3588 ubuntu22

- Kernel version
- 内核版本
5.10.160

**Additional context**
**附加说明**
log:
04-28 22:37:06.404   252  3559 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.bilibili.azurlane/com.manjuu.azurlane.MainActivity bnds=[16,174][188,414]} from uid 10073
04-28 22:37:06.415   252  3559 V SplashScreenExceptionList: SplashScreen checking exception for package com.bilibili.azurlane (target sdk:30) -> false
04-28 22:37:06.465   252   281 I ActivityManager: Start proc 7203:com.bilibili.azurlane/u0a91 for next-top-activity {com.bilibili.azurlane/com.manjuu.azurlane.MainActivity}
04-28 22:37:06.471   252  3899 D CoreBackPreview: Window{e32e6b7 u0 Splash Screen com.bilibili.azurlane}: Setting back callback OnBackInvokedCallbackInfo{mCallback=android.window.IOnBackInvokedCallback$Stub$Proxy@84db589, mPriority=0}
04-28 22:37:06.518  7203  7203 V GraphicsEnvironment: ANGLE Developer option for 'com.bilibili.azurlane' set to: 'default'
04-28 22:37:06.518  7203  7203 V GraphicsEnvironment: ANGLE GameManagerService for com.bilibili.azurlane: false
04-28 22:37:06.557   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.573   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.590   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.606   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.623   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.640   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.656   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.673   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.690   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.706   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.723   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.740   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.756   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.773   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.790   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.806   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.823   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.840   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.856   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.873   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.890   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.906   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.923   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.940   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.957   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.973   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:06.990   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:07.006   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:07.023   114   114 E CompositionEngine: [Splash Screen com.bilibili.azurlane#679] Invalid device requested composition type change: Client (1) --> Device (2)
04-28 22:37:07.644   252   718 D CoreBackPreview: Window{d8ea625 u0 com.bilibili.azurlane/com.manjuu.azurlane.MainActivity}: Setting back callback OnBackInvokedCallbackInfo{mCallback=android.window.IOnBackInvokedCallback$Stub$Proxy@ba7dab, mPriority=0}
04-28 22:37:07.682   252   270 I ActivityTaskManager: Displayed com.bilibili.azurlane/com.manjuu.azurlane.MainActivity: +1s270ms
04-28 22:37:07.691   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.707   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.724   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.740   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.757   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.774   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.790   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.807   114   114 E CompositionEngine: [Background for SurfaceView[com.bilibili.azurlane/com.manjuu.azurlane.MainActivity]#692] Invalid device requested composition type change: SolidColor (3) --> Device (2)
04-28 22:37:07.813  7203  7228 I Unity   : ApplicationInfo com.bilibili.azurlane version 7.1.1 build 04053956-3199-4f9d-a2a6-fae11168be50
[... cut ...]
04-28 22:37:08.284   252   718 I WindowManager: WIN DEATH: Window{d8ea625 u0 com.bilibili.azurlane/com.manjuu.azurlane.MainActivity}
04-28 22:37:08.284   252   718 W InputManager-JNI: Input channel object 'd8ea625 com.bilibili.azurlane/com.manjuu.azurlane.MainActivity (client)' was disposed without first being removed with the input manager!
04-28 22:37:08.287   252  2928 W ActivityTaskManager: Force removing ActivityRecord{baba7db u0 com.bilibili.azurlane/com.manjuu.azurlane.MainActivity} t15}: app died, no saved state

CNflysky, 2024-04-29:
seems ok……？
貌似……没问题？
[image]

CoronaAustralis, 2024-04-30:
> seems ok……？ 貌似……没问题？ ![Screenshot_20240429-201157.png](https://private-user-images.githubusercontent.com/48781081/326431836-5a3830a7-ceef-4483-bd58-464b3eeea871.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0NDUwODcsIm5iZiI6MTcxNDQ0NDc4NywicGF0aCI6Ii80ODc4MTA4MS8zMjY0MzE4MzYtNWEzODMwYTctY2VlZi00NDgzLWJkNTgtNDY0YjNlZWVhODcxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDAyMzk0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRmZDFlMDQwMDU3YzgxNWE2MDUzZTA2OWJkMmNmN2VhMTJmNDgzMTkxZmU5ZWNiYTVjOWNjZTJiYjBiOTkyYjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m8qJNMQotLPB690n-Mu8nPSSVuLZxiOd-0cKq8T54W8)

奇怪，是我系统的问题？我使用的是[Joshua-Riek](https://github.com/Joshua-Riek/ubuntu-rockchip)
的ubuntu22.04 server版，但是这个系统也是panfork驱动，是panfork版本的问题吗？我改成armbian试一下吧

CNflysky, 2024-04-30:
> 奇怪，是我系统的问题？我使用的是[Joshua-Riek](https://github.com/Joshua-Riek/ubuntu-rockchip)
> 的ubuntu22.04 server版，但是这个系统也是panfork驱动，是panfork版本的问题吗？我改成armbian试一下吧

well, `panfork` actually is a userspace gpu driver，redroid will use its own closed-source mali gpu driver(`libGLESv2_mali.so`), so maybe it was nothing to do with `panfork` driver.

btw i use `bookworm server` version of armbian which doesn't contain a userspace gpu driver at all. 

额，panfork是用户空间的驱动库，redroid容器本身会用自己的闭源mali gpu驱动(`libGLESv2_mali.so`)，所以应该和panfork驱动没有关系。
而且我用的armbian bookworm server版系统是没有用户空间GPU驱动的。

CoronaAustralis, 2024-04-30:
我是菜鸟😂，所以你的意思是redroid的libGLESv2_mali.so将直接与内核gpu驱动交互，而内核gpu驱动则应该是在系统镜像构建时一起被编译进去。这时宿主机的的用户空间gpu驱动是什么，有或没有，根本没影响。

CoronaAustralis, 2024-05-06:
官服，但是`小米`服我也试了，还是闪退，会是投屏软件的问题吗？我使用的是scrcpy投屏。

lonzzi, 2024-05-18:
系统信息：
orangepi5
rk3588s
Armbian 24.2.1 Jammy
Kernel: 5.10.160-legacy-rk35xx

我这边测试也是初次打开 碧蓝航线 有同意条款之后就闪退，后续再打开都是直接闪退。
Android 13 和 Android 12 都测试过结果也都是一样的。

SeeFlowerX, 2024-06-07:
如果scrcpy不是最新版，你们可以试一下升级到最新版

GitProPlusMaxUltra, 2024-07-08:
太难了，米哈游的几款，全是要进入游戏后下载数据，下载过程中就会报错重启好几次；
下载之后的检验完整性到一定进度就会触发重启：
`07-08 10:03:30.284  3511  3511 W SetupWizard: [bty] getAssistantOptInState() component does not exist
07-08 10:03:30.311  3511  3511 D FakeWiFi: fake wifi info returned
07-08 10:03:30.321  3511  3511 W SetupWizard: [bty] getAssistantOptInState() component does not exist
07-08 10:03:30.324  3511  3511 I SetupWizard: [DeferredSetupNotificationSchedulerService] Deferred notification scheduled at Tue Jul 09 09:00:00 GMT 2024
07-08 10:03:30.324  3511  3511 D CompatibilityChangeReporter: Compat change id reported: 194532703; UID 10074; state: DISABLED
07-08 10:03:30.325   250   683 D CompatibilityChangeReporter: Compat change id reported: 194532703; UID 10074; state: DISABLED
07-08 10:03:31.184    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 3577/3579 (recycle/alloc) - 2/3578 (fetch/transfer)
07-08 10:03:31.909    95  2935 I C2RKMpiEnc: IDR frame produced
07-08 10:03:36.212   250   279 D OomAdjuster: Not killing cached processes
07-08 10:03:36.349    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 3732/3734 (recycle/alloc) - 2/3733 (fetch/transfer)
07-08 10:03:36.859  2904  2904 D FakeWiFi: fake wifi info returned
07-08 10:03:37.136  1173  1173 D BoundBrokerSvc: onBind: Intent { act=com.google.android.gms.phenotype.service.START pkg=com.google.android.gms }
07-08 10:03:37.136  1173  1173 D BoundBrokerSvc: Loading bound service for intent: Intent { act=com.google.android.gms.phenotype.service.START pkg=com.google.android.gms }
07-08 10:03:37.140  1173  1173 E .gms.persistent: Failed to open APK '/data/user_de/0/com.google.android.gms/files/locale-filtered-resources-943390677-zh.zip': Invalid file
07-08 10:03:37.141  1173  1173 W BaseAppContext: ResourcesHelper: java.lang.RuntimeException: addAssetPath failed: 0 for /data/user_de/0/com.google.android.gms/files/locale-filtered-resources-943390677-zh.zip
07-08 10:03:37.161  1173  2722 D FakeWiFi: fake wifi info returned
07-08 10:03:37.170  1173  2722 D FakeWiFi: fake wifi info returned
07-08 10:03:37.174  3511  3511 V ClearcutMetricXmitter: Transmission is done.
07-08 10:03:37.176  1173  2722 D FakeWiFi: fake wifi info returned
07-08 10:03:37.633  1831  2063 D TrafficStats: tagSocket(84) with statsTag=0x8001, statsUid=-1
07-08 10:03:39.576   630   805 E TcpSocketTracker: Expect to get family 10 SOCK_DIAG_BY_FAMILY message but get 2
07-08 10:03:39.577   630   805 E TcpSocketTracker: Expect to get family 2 SOCK_DIAG_BY_FAMILY message but get 2
07-08 10:03:41.549    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 3888/3890 (recycle/alloc) - 2/3889 (fetch/transfer)
07-08 10:03:44.163   250  2566 D OomAdjuster: Not killing cached processes
07-08 10:03:45.039   179   191 E Netd    : Failed to dump IPv4 sockets struct tcp_info: No such file or directory
07-08 10:03:45.039   179   191 E TcpSocketMonitor: Failed to poll TCP socket info: No such file or directory07-08 10:03:46.691    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 4042/4044 (recycle/alloc) - 2/4043 (fetch/transfer)
07-08 10:03:47.019  2904  2929 D TrafficStats: tagSocket(198) with statsTag=0xffffffff, statsUid=-1
07-08 10:03:47.653  1831  2063 D TrafficStats: tagSocket(84) with statsTag=0x8001, statsUid=-1
07-08 10:03:51.474  2904  2904 D CompatibilityChangeReporter: Compat change id reported: 160794467; UID 10092; state: ENABLED
07-08 10:03:51.480   250  1024 D OomAdjuster: Not killing cached processes
07-08 10:03:51.491  2904  2904 D FakeWiFi: fake wifi info returned
07-08 10:03:51.620  2904  2904 D FakeWiFi: fake wifi info returned
07-08 10:03:51.640  2904  2915 W System  : A resource failed to call close.
07-08 10:03:51.641  2904  2915 W System  : A resource failed to call close.
07-08 10:03:51.841    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 4193/4195 (recycle/alloc) - 2/4195 (fetch/transfer)
07-08 10:03:52.162  1173  1173 D BoundBrokerSvc: onUnbind: Intent { act=com.google.android.gms.phenotype.service.START pkg=com.google.android.gms }
07-08 10:03:52.174    95  2935 I C2RKMpiEnc: IDR frame produced
07-08 10:03:53.110  1348  2261 D TrafficStats: tagSocket(103) with statsTag=0x20003000, statsUid=-1
07-08 10:03:56.886   109   133 W TransactionTracing: Could not find layer id -1
07-08 10:03:56.966    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 4338/4340 (recycle/alloc) - 2/4339 (fetch/transfer)
07-08 10:03:57.344  2904  2929 D TrafficStats: tagSocket(328) with statsTag=0xffffffff, statsUid=-1
07-08 10:03:57.661  1831  2063 D TrafficStats: tagSocket(84) with statsTag=0x8001, statsUid=-1
07-08 10:03:58.426  2904  2923 D TrafficStats: tagSocket(209) with statsTag=0xffffffff, statsUid=-1
07-08 10:03:59.580   630   805 E TcpSocketTracker: Expect to get family 10 SOCK_DIAG_BY_FAMILY message but get 2
07-08 10:03:59.580   630   805 E TcpSocketTracker: Expect to get family 2 SOCK_DIAG_BY_FAMILY message but get 2
[... cut ...]
07-08 10:04:07.753   109   133 W TransactionTracing: Could not find layer id -1
07-08 10:04:07.920   109   133 W TransactionTracing: Could not find layer id -1
07-08 10:04:09.577  1831  1907 D TrafficStats: tagSocket(77) with statsTag=0x6900, statsUid=-1
07-08 10:04:12.291    95  2935 D BufferPoolAccessor2.0: bufferpool2 0xe9ec0808 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 4798/4800 (recycle/alloc) - 2/4799 (fetch/transfer)
07-08 10:04:12.324    95  2935 I C2RKMpiEnc: IDR frame produced`
我用最新的 13 版本运行，似乎内存问题导致的容器重启

CNflysky, 2024-07-08:
> 太难了，米哈游的几款，全是要进入游戏后下载数据，下载过程中就会报错重启好几次；
> 下载之后的检验完整性到一定进度就会触发重启：
> `07-08 10:03:30.284  3511  3511 W SetupWizard: [bty] getAssistantOptInState() component does not exist
> 07-08 10:03:30.311  3511  3511 D [...]

please provide more information (board model, host os info, memory/disk info etc.)

GitProPlusMaxUltra, 2024-07-09:
- Required Information
基本信息

-Hardware Information
硬件信息
OrangePi 5 Plus，CPU RK3588，Memory 32GB，Disk emmc 256GB

- Docker Image
容器镜像
13.0.0-latest

- OS Information(Debian、Ubuntu……)
系统信息
Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal

- Kernel version
内核版本
5.10.160-legacy-rk35xx

GitProPlusMaxUltra, 2024-07-09:
今天下午突然回头看了看当时参考着安装用户空间驱动的项目[malior](https://github.com/ChisBread/malior)，本身也是一个虚拟安卓提供[容器镜像](docker pull chisbread/rk3588-gaming:redroid-firefly)，我发现这个项目的适配似乎特别完整。

可以通过它提供的镜像运行 3Dmark 的检测项目：
[Sling Shot Extreme](https://github.com/CNflysky/redroid-rk3588/assets/126841763/c0e96d6f-c2a1-40d8-902a-dc1e7a5b13f7)
且能达到极限值，估计跑米哈游相关的也不是问题。

在使用当前项目的 12/13 最新容器镜像或者之前的版本时，无论是米哈游游戏还是3Dmark都会触发重启。

希望开发者可以话能够依据 malior 项目 修改完善本项目的镜像，malior 它没预置 谷歌全家桶 也没有虚拟WiFi，界面看着也怪怪的不是很好看 `(*>﹏<*)′

SeeleVolleri, 2024-07-09:
>资源下载完后开始检测完整性，检测到一定进度就触发重启，整个开发板系统也跟着重启了，至此无法开始游戏，检测完整性到一定进度就重启。

你设备怎么供电的？一般情况下板子不会无缘无故重启。症状貌似表明你在使用12v1a等小于20瓦的电源。

KylinDemons, 2024-07-09:
> * **Required Information**
>   基本信息
>   Docker version 24.0.9
> * **Hardware Information**
>   硬件信息
>   OrangePi 5 Plus，CPU RK3588，Memory 32GB，Disk emmc 256GB
> * **Docker Image**
>   容器镜像
>   13.0.0-latest
> * **OS Information(Debian、Ubuntu……)**
>   系统信息
>   Armbian_24.2.3_Orangepi5-plus_bookworm_legacy_5.10.160_minimal
> * **Kernel version**
>   内核版本
>   5.10.160-legacy-rk35xx
> 
> 运行其它的一些类似米哈游这样架构的大型游戏，就是说客户端不大，进入游戏后下载大量资源，下载资源中间不规则触发几次系统重启，资源下载完后开始检测完整性，检测到一定进度就触发重启，整个开发板系统也跟着重启了，至此无法开始游戏，检测完整性到一定进度就重启。 Run some other large games similar to the structure of Mihayo, that is to say, the client is not big, after entering the game to download a large number of resources, download resources in the middle of the irregular triggered a number of system reboot, the resources after the download began to detect the integrity of the detection to a certain progress triggered a reboot, the entire development board system followed the reboot to the inability to start the game, the detection of the integrity of the progress of a certain restart.

请检查供电问题，不要使用PD协议的电源。电源必须标明支持5V/4A。

GitProPlusMaxUltra, 2024-07-09:
并非电源问题
Not a power supply problem

GitProPlusMaxUltra, 2024-07-09:
_有点懵，我刚刚也重新创建运行了下13的容器，3Dmark 连着测试两项
[image]

这次 3Dmark 没触发重启，接着就安装 taptap，通过它下载 绝区零，安装游戏后打开，下载数据进度条到17-18%重启了...

- 重启时的一些 adb logcat 日志：
`
07-09 09:42:00.000    95   206 D AudioFlinger: mixer(0xb400f1563fed09a0) throttle end: throttle time(17)
07-09 09:42:00.600    95   206 D AudioFlinger: mixer(0xb400f1563fed09a0) throttle end: throttle time(17)
07-09 09:42:01.852    95   206 D AudioFlinger: mixer(0xb400f1563fed09a0) throttle end: throttle time(1)
07-09 09:42:01.960  5437  5473 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:03.661    95   206 D AudioFlinger: mixer(0xb400f1563fed09a0) throttle end: throttle time(4)
07-09 09:42:05.135  1148  4187 I NetworkScheduler.Stats: Task com.google.android.gms/com.google.android.gms.auth.account.be.legacy.AuthCronService started execution. cause:4 exec_start_elapsed_seconds: 1516 [CONTEXT service_id=218 ]
07-09 09:42:05.142  1148  1939 I NetworkScheduler.Stats: Task com.google.android.gms/com.google.android.gms.auth.account.be.legacy.AuthCronService finished executing. cause:4 result: 1 elapsed_millis: 33 uptime_millis: 33 exec_start_elapsed_seconds: 1516 [CONTEXT service_id=218 ]
07-09 09:42:05.146    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 7650/7652 (recycle/alloc) - 5/7651 (fetch/transfer)
07-09 09:42:09.962  3744  3783 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:10.130    91  7861 I C2RKMpiEnc: IDR frame produced
07-09 09:42:10.264    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 7803/7805 (recycle/alloc) - 5/7804 (fetch/transfer)
07-09 09:42:10.699    95   206 D AudioFlinger: mixer(0xb400f1563fed09a0) throttle end: throttle time(5)
07-09 09:42:12.009  5437  5473 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:12.928  7815  7846 D TrafficStats: tagSocket(202) with statsTag=0xffffffff, statsUid=-1
07-09 09:42:13.053   987  1348 E TcpSocketTracker: Expect to get family 10 SOCK_DIAG_BY_FAMILY message but get 2
07-09 09:42:13.053   987  1348 E TcpSocketTracker: Expect to get family 2 SOCK_DIAG_BY_FAMILY message but get 2
07-09 09:42:15.413    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 7957/7959 (recycle/alloc) - 5/7958 (fetch/transfer)
07-09 09:42:16.241   147   172 E Netd    : Failed to dump IPv4 sockets struct tcp_info: No such file or directory
07-09 09:42:16.241   147   172 E TcpSocketMonitor: Failed to poll TCP socket info: No such file or directory
07-09 09:42:20.027  3744  3783 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:20.580    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 8112/8114 (recycle/alloc) - 5/8113 (fetch/transfer)
07-09 09:42:22.055  5437  5473 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:24.652  1338  8827 D TrafficStats: tagSocket(83) with statsTag=0x20004901, statsUid=10052
07-09 09:42:25.705    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 8265/8267 (recycle/alloc) - 5/8266 (fetch/transfer)
07-09 09:42:28.066  1338  8853 D TrafficStats: tagSocket(74) with statsTag=0x20002500, statsUid=-1
07-09 09:42:28.094  1338  8853 W Conscrypt: Could not set socket write timeout: java.net.SocketException: Socket closed
07-09 09:42:28.095  1338  8853 W Conscrypt:     at com.google.android.gms.org.conscrypt.Platform.setSocketWriteTimeout(:com.google.android.gms@214815044@21.48.15 (190400-414534850):2)
07-09 09:42:28.095  1338  8853 W Conscrypt:     at com.google.android.gms.org.conscrypt.ConscryptFileDescriptorSocket.setSoWriteTimeout(:com.google.android.gms@214815044@21.48.15 (190400-414534850):0)
07-09 09:42:28.354  1338  8853 V NativeCrypto: SSL handshake aborted: ssl=0xb400fcb6ddb59b18: I/O error during system call, Connection reset by peer
07-09 09:42:28.374  1338  8676 D TrafficStats: tagSocket(163) with statsTag=0x20002500, statsUid=-1
07-09 09:42:30.061  3744  3783 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:30.177    91  7861 I C2RKMpiEnc: IDR frame produced
07-09 09:42:30.596  6650  6731 I tnet-jni: 78 - [spdyPingRecvCallback] -
07-09 09:42:30.596  6650  6731 I tnet-jni: 78 - [SpdySessionCallBack.spdyPingRecvCallback] - org.android.spdy.SpdySession@c53c2f7
07-09 09:42:30.873    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 8420/8422 (recycle/alloc) - 5/8421 (fetch/transfer)
07-09 09:42:32.058  5437  5473 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
07-09 09:42:33.058   987  1348 E TcpSocketTracker: Expect to get family 10 SOCK_DIAG_BY_FAMILY message but get 2
07-09 09:42:33.059   987  1348 E TcpSocketTracker: Expect to get family 2 SOCK_DIAG_BY_FAMILY message but get 2
07-09 09:42:36.031    91  7861 D BufferPoolAccessor2.0: bufferpool2 0xf1ec33e8 : 2(1323008 size) total buffers - 1(661504 size) used buffers - 8575/8577 (recycle/alloc) - 5/8576 (fetch/transfer)
07-09 09:42:38.366  5712  8874 D TrafficStats: tagSocket(183) with statsTag=0xffffffff, statsUid=-1
07-09 09:42:40.099  3744  3783 D TrafficStats: tagSocket(5) with statsTag=0x8001, statsUid=-1
[... cut ...]
     -v ./data:/data \
     -p 5555:5555 \
     --name redroid \
     redroid-rk3588:13.0.0-latest \
     androidboot.use_memfd=false \
     androidboot.redroid_fps=120 \
     androidboot.redroid_dpi=320 \
     androidboot.redroid_width=720 \
     androidboot.redroid_height=1280 \
     androidboot.redroid_net_ndns=1 \
     androidboot.redroid_net_dns1=223.5.5.5 \
     androidboot.redroid_net_dns2=8.8.8.8 \
     androidboot.redroid_gpu_mode=mail \
     --mount=type=bind, source=/dev/mali0, destination=/dev/mali0

KylinDemons, 2024-07-09:
> _有点懵，我刚刚也重新创建运行了下13的容器，3Dmark 连着测试两项 [...]

目前测试碧蓝航线确实会闪退，报错和你一样，正在排查内核问题

KylinDemons, 2024-07-09:
@CoronaAustralis 方便的话进群交流，目前在排查这个问题

theone-wu, 2024-07-12:
> 目前测试碧蓝航线确实会闪退，报错和你一样，正在排查内核问题

目前有个快递的app开始还能正常进去 后面 不给位置授权 一直就崩溃
是不是位置授权还是环境检测不通过 有些app会检测设备的 这个希望有人可以做个 出来 模拟硬件信息的

theone-wu, 2024-07-12:
目前有个快递的app开始还能正常进去 后面 不给位置授权 一直就崩溃
是不是位置授权还是环境检测不通过 有些app会检测设备的 这个希望有人可以做个 出来 模拟硬件信息的

CNflysky, 2024-07-12:
enable `CONFIG_ARM64_VA_BITS=39` in your kernel:
`make menuconfig` -> `Virtual address space size` -> `39`.
later i 'll upload a modified kernel package based on `armbian legacy kernel` config.

在你的内核中启用`CONFIG_ARM64_VA_BITS=39`:
`make menuconfig` -> `Virtual address space size` -> `39`.
稍后我会上传一份基于`armbian legacy内核`配置修改过的内核包。

closed 2024-07-12 by CNflysky

CNflysky, 2024-07-13:
[Uploading linux-image-5.10.160-rockchip-rk3588_5.10.160-rockchip-rk3588-1_arm64.zip…](kernel)
[Uploading linux-headers-5.10.160-rockchip-rk3588_5.10.160-rockchip-rk3588-1_arm64.zip…](headers)
github can not upload .deb files so i have zipped those files
tested on armbian ubuntu 22.04 (shipped with panfork driver)

CNflysky, 2024-07-14:
[kernel](https://github.com/user-attachments/files/16227927/linux-image-5.10.160-rockchip-rk3588_5.10.160-rockchip-rk3588-1_arm64.zip)  
[headers](https://github.com/user-attachments/files/16227929/linux-headers-5.10.160-rockchip-rk3588_5.10.160-rockchip-rk3588-1_arm64.zip)
uninstall armbian kernel, unzip archive, install deb file, reboot.
btw i've bumped rknpu driver version to 0.9.6 so you can use this kernel to run llms.

GitProPlusMaxUltra, 2024-07-15:
芜湖~感谢各位开发者！
wuhu~Thanks to the developers!