CNflysky/redroid-rk3588 #5: vpn不能用吗？ / VPN support?

minquejoe, 2024-05-04:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
**描述Bug**
the combinations of following setting are tried while creating container, but the proxy function seems didn't work.
尝试在创建容器时添加了下面两个设置的排列组合，试了两个游戏加速器和Clash for Android均不能正常代理。
`-v /dev/net/tun:/dev/tun`
`--sysctl net.ipv6.conf.all.disable_ipv6=0`

**Required Information**
**基本信息**
- container Image name
- 镜像名称
`cnflysky/redroid-rk3588:12.0.0-latest`

- OS Information
- 系统信息
orangepi official ubuntu image
orangepi官方的ubuntu
```
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
```

- Kernel version
- 内核版本
`Linux orangepi5 5.10.160-rockchip-rk3588 #1.1.8 SMP Mon Nov 13 10:15:33 CST 2023 aarch64 aarch64 aarch64 GNU/Linux`

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
以下是复现该问题的步骤：
试了雷神加速器，提示加速失败；shimakazego直接闪退；CFA提示cannot create interface。

**Screenshots**
**截图**
If applicable, add screenshots to help explain your problem.
如果可以，添加一些截图以更好地解释问题

**Additional context**
**附加说明**
Add any other context about the problem here.
logcat messages are suggested place into pastebin.
如果有额外的问题说明，请放在这里
logcat信息建议放在pastebin里
CFA运行时的logcat：
```
05-04 14:54:01.340  5316  5338 D ClashForAndroid: ClashRuntime: initialize
05-04 14:54:01.359  5316  5493 D ClashForAndroid: CloseModule: initialize
05-04 14:54:01.359  5316  5339 D ClashForAndroid: TunModule: initialize
05-04 14:54:01.360  5316  5521 D ClashForAndroid: ConfigurationModule: initialize
05-04 14:54:01.365  5316  5521 D ClashForAndroid: SideloadDatabaseModule: initialize
05-04 14:54:01.365  5316  5493 D ClashForAndroid: NetworkObserveModule: initialize
05-04 14:54:01.366  5316  5339 D ClashForAndroid: DynamicNotificationModule: initialize
05-04 14:54:01.366  5316  5340 D ClashForAndroid: AppListCacheModule: initialize
05-04 14:54:01.366  5316  5496 D ClashForAndroid: TimeZoneModule: initialize
05-04 14:54:01.368   228   567 D ConnectivityService: NetReassign [no changes]
05-04 14:54:01.371  5316  5493 D ClashForAndroid: SuspendModule: initialize
05-04 14:54:01.374  5316  5496 D ClashForAndroid: DNS: [8.8.8.8:53]
05-04 14:54:01.374  5316  5496 D ClashForAndroid: Network: 100 of [100]
05-04 14:54:01.374  5316  5496 D ClashForAndroid: DNS: [8.8.8.8:53]
05-04 14:54:01.379  5316  5340 D ClashForAndroid: Installed 92 packages cached
05-04 14:54:01.386   228   781 E VpnJni  : Cannot allocate TUN: Bad file descriptor
05-04 14:54:01.387  5316  5338 E ClashForAndroid: Create clash runtime: Cannot create interface
05-04 14:54:01.387  5316  5338 E ClashForAndroid: java.lang.IllegalStateException: Cannot create interface
05-04 14:54:01.387  5316  5338 E ClashForAndroid: 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2433)
05-04 14:54:01.387  5316  5338 E ClashForAndroid: 	at android.os.Parcel.createException(Parcel.java:2409)
[... 64 lines cut ...]
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   - WORKGROUP
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   default-nameserver:
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   - 223.5.5.5
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   - 119.29.29.29
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   - 8.8.4.4
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   - 1.0.0.1
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   nameserver-policy: {}
05-04 14:54:01.506  5316  5347 I ClashForAndroid:   
05-04 14:54:01.508  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.508  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.508  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.508  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.508  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.509  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.509  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.509  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.509  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.509  5316  5349 I ClashForAndroid: Start initial compatible provider XXX
05-04 14:54:01.525  5316  5452 I ClashForAndroid: HTTP proxy listening at: 127.0.0.1:7890
05-04 14:54:01.525  5316  5452 I ClashForAndroid: SOCKS proxy listening at: 127.0.0.1:7891
05-04 14:54:01.545   228   228 I NotificationService: Cannot find enqueued record for key: 0|com.github.kr328.clash|2131296568|null|10084
05-04 14:54:01.593   228   228 I NotificationService: Cannot find enqueued record for key: 0|com.github.kr328.clash|2131296568|null|10084
05-04 14:54:01.636  3040  3062 D BufferPoolAccessor2.0: bufferpool2 0xb40000747a3fdcb8 : 5(19200 size) total buffers - 4(15360 size) used buffers - 167470/167475 (recycle/alloc) - 173/334941 (fetch/transfer)
05-04 14:54:02.209   228   278 D AutofillManagerService: onBackKeyPressed()
```

CNflysky, 2024-05-04:
try `modprobe tun`.
试试`modprobe tun`

minquejoe, 2024-05-05:
perfect!
问题解决。

closed 2024-05-05 by minquejoe