CNflysky/redroid-rk3588 #7: 玩游戏一直断网

XiaoMing-Fish, 2024-07-09:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
**描述Bug**
玩游戏一直断网 

**Required Information**
**基本信息**
- docker image name
- 镜像名称
e.g. `12.0.0-latest`

- Board Information
- 板卡信息
e.g. `Orange Pi 5 Plus + 16 GB RAM`

- OS Information
- 系统信息
e.g. `Armbian Bookworm Server`
Linux orangepi5plus 5.10.160-rockchip-rk3588 #1.0.10 SMP Fri Jun 28 10:49:33 CST 2024 aarch64 GNU/Linux
- Kernel version
- 内核版本
5.10.160-rockchip-rk3588

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
以下是复现该问题的步骤：
打开梦幻西游手游。网段断断续续

**Screenshots**
**截图**
无

**Additional context**
**附加说明**
希望添加h256解码 ，是否有控制cpu和内存限制的参数
logcat信息建议放在pastebin里

CNflysky, 2024-07-09:
try specify dns:
```
androidboot.redroid_net_ndns=1
androidboot.redroid_net_dns1=114.114.114.114
```
`h265 decoding` seems ok in redroid.
resource quota can be set with `--cpu` / `--memory` arguments.

XiaoMing-Fish, 2024-07-22:
> try specify dns:
> 
> ```
> androidboot.redroid_net_ndns=1
> androidboot.redroid_net_dns1=114.114.114.114
> ```
> dns已经修复
> `h265 decoding` seems ok in redroid. resource quota can be set with `--cpu` / `--memory` arguments.
 `--cpu` / `--memory`  无法限制容器占用硬件资源

closed 2024-08-23 by CNflysky