CNflysky/redroid-rk3588 #17: 请问怎么指定设备型号、厂商之类的

Coldairarrow, 2024-08-02:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
**描述Bug**
A clear and concise description of what the bug is.
提供一个清晰简洁的问题描述
请问怎么指定设备型号、厂商之类的

**Required Information**
**基本信息**
- docker image name
- 镜像名称
e.g. `12.0.0-latest`

- Board Information
- 板卡信息
e.g. `Orange Pi 5 Plus + 16 GB RAM`

- OS Information
- 系统信息
e.g. `ubuntu 22`

- Kernel version
- 内核版本
e.g. `5.10.160-rockchip-rk35xx`

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
以下是复现该问题的步骤：

**Screenshots**
**截图**
If applicable, add screenshots to help explain your problem.
如果可以，添加一些截图以更好地解释问题

**Additional context**
**附加说明**
Add any other context about the problem here.
logcat messages are suggested place into pastebin.
如果有额外的问题说明，请放在这里
logcat信息建议放在pastebin里

SeeleVolleri, 2024-08-02:
直接将`ro.product.model`、`ro.product.brand`等`build.prop`参数像README中的示例一样添加到末尾即可

Coldairarrow, 2024-08-02:
[image]
[image]

加了没效果呀

Coldairarrow, 2024-08-02:
> 直接将`ro.product.model`、`ro.product.brand`等`build.prop`参数像README中的示例一样添加到末尾即可

大哥麻烦解惑下

Coldairarrow, 2024-08-03:
搞定了，通过挂载/system/build.prop 实现了

closed 2024-08-03 by Coldairarrow

scjtqs2, 2025-05-24:
> 挂载是怎么实现的，方便说下么
> 
> > 搞定了，通过挂载/system/build.prop 实现了

从docker镜像内部拷贝/system/build.prop 到本地。然后贴到chatgpt，让它帮你改，改好了，用-v提换掉/system/build.prop文件

yigger, 2025-06-15:
> > 挂载是怎么实现的，方便说下么
> > > 搞定了，通过挂载/system/build.prop 实现了
> 
> 从docker镜像内部拷贝/system/build.prop 到本地。然后贴到chatgpt，让它帮你改，改好了，用-v提换掉/system/build.prop文件

修改替换以后似乎启动不了。