CNflysky/redroid-rk3588 #18: 能否添加对RK3568的支持？

li-chx, 2024-08-16:
**Describe the bug**
**描述Bug**
RK3568 docker container启动不成功，持续报错循环
期望redroid-rk3588 在 redroid-rk3568上直接能用并不现实，如果开发者不愿付出额外的精力添加支持，请直接关掉该issue。
为我的打扰致歉。

**Required Information**
**基本信息**
- docker image name
- 镜像名称
`12.0.0-latest`
13.0.0正在试，一会儿补充

- Board Information
- 板卡信息
`ATOM-pi RK3568 + 4 GB RAM`
反正就是后悔，资料太少

- OS Information
- 系统信息
```
lichx@atompi-ca1:~/redroid-rk3588$ hostnamectl
 Static hostname: atompi-ca1
       Icon name: computer
      Machine ID: 3f2681a1f3f44b8681c5a1715b99b7b0
         Boot ID: ff9eccbfa52344b6a355843f243e7481
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 5.10.160
    Architecture: arm64
```

- Kernel version
- 内核版本
- `5.10.160-rockchip-rk3568`
```
lichx@atompi-ca1:~/redroid-rk3588$ uname -a
Linux atompi-ca1 5.10.160 #19 SMP Fri Aug 16 18:11:01 CST 2024 aarch64 GNU/Linux
```

Steps to reproduce the behavior:
以下是复现该问题的步骤：
直接在RK3568上clone项目，检查依赖，然后`docker compose up -d`

**Additional context**
**附加说明**
[PasteBin地址 logcat 信息](https://pastebin.com/6ei7j2Rp)
[Serial Port log](https://pastebin.com/G8jEYsn6)
RK3568与RK3588使用不同的Mali GPU，但是我发现buildroot下mali_csffw.bin的sha256是一致的。

yyzz521, 2024-08-16:
对不起，做不到

li-chx, 2024-08-16:
没事 打扰了
已经在考虑加钱上rk3588s了 这个太坐牢了

closed 2024-08-16 by li-chx