CNflysky/redroid-rk3588 #20: 功能期望

XiaoMing-Fish, 2024-08-21:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
**描述Bug**
A clear and concise description of what the bug is.
提供一个清晰简洁的问题描述
是否添加虚拟电池
是否能代理ip
能否集成改机
**Required Information**
**基本信息**
- docker image name
- 镜像名称
e.g. `13.0.0-latest`
12.0.0-latest
- Board Information
- 板卡信息
e.g. `Orange Pi 5 Plus + 16 GB RAM`
Orange Pi 5 Plus + 16 GB RAM
- OS Information
- 系统信息
e.g. `Armbian Bookworm Server`
Debian 11 (bullseye)
- Kernel version
- 内核版本
e.g. `5.10.160-rockchip-rk35xx`
5.10.160-rockchip-rk3588

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
以下是复现该问题的步骤：

**Screenshots**
**截图**
If applicable, add screenshots to help explain your problem.
如果可以，添加一些截图以更好地解释问题

**Additional context**
**附加说明**
Add any other context about the problem here.
logcat messages are suggested place into pastebin.
如果有额外的问题说明，请放在这里
logcat信息建议放在pastebin里

CNflysky, 2024-08-21:
虚拟电池可以通过在/sys下放置一些文件来实现（`/sys/power`）
不太清楚你说的代理IP是什么，如果是VPN的话，已经支持了
修改设备信息可以通过挂载prop的方式实现

XiaoMing-Fish, 2024-08-21:
vpn这个没看到设置的属性
ro修改prop视乎无效
电池这个我会试试看
tks

XiaoMing-Fish, 2024-08-21:
设备再用第三方代理apk会闪退，例如奇游加速器

CNflysky, 2024-08-21:
> 设备再用第三方代理apk会闪退，例如奇游加速器

请提供日志，须知「在没有日志的情况下就诊断问题无异于闭眼开车」

CNflysky, 2024-08-21:
> vpn这个没看到设置的属性
> ro修改prop视乎无效
> 电池这个我会试试看
> tks

#5

SeeleVolleri, 2024-08-21:
modprobe tun

SeeleVolleri, 2024-08-21:
> this logcat
> 08-21 03:18:32.707 237 237 W NotificationHistory: Attempted to add notif for locked/gone/disabled user 0
> 08-21 03:18:32.708 1300 1300 D NotificationObserver: onNotification: []; EventType: TYPE_NOTIFICATION_STATE_CHANGED; EventTime: 1388797929; PackageName: fun.kitsunebi.kitsunebi4android; MovementGranularity: 0; Action: 0; ContentChangeTypes: []; WindowChangeTypes: [] [ ClassName: android.app.Notification; Text: []; ContentDescription: null; ItemCount: -1; CurrentItemIndex: -1; Enabled: false; Password: false; Checked: false; FullScreen: false; Scrollable: false; BeforeText: null; FromIndex: -1; ToIndex: -1; ScrollX: 0; ScrollY: 0; MaxScrollX: 0; MaxScrollY: 0; ScrollDeltaX: -1; ScrollDeltaY: -1; AddedCount: -1; RemovedCount: -1; ParcelableData: Notification(channel=Kitsunebi shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xe2 color=0x00000000 actions=1 vis=PRIVATE) ]; recordCount: 0
> 08-21 03:18:32.709 122 46843 I adbd : host-71: read thread spawning
> 08-21 03:18:32.709 122 46844 I adbd : host-71: write thread spawning
> 08-21 03:18:32.709 122 46843 I adbd : host-71: read failed: Success
> 08-21 03:18:32.709 122 46843 I adbd : host-71: connection terminated: read failed
> 08-21 03:18:32.709 122 122 I adbd : host-71: already offline
> 08-21 03:18:32.709 122 122 I adbd : destroying transport host-71
> 08-21 03:18:32.709 122 122 I adbd : BlockingConnectionAdapter(host-71): stopping
> 08-21 03:18:32.710 407 407 W ShadeListBuilder: (Build 19996) Duplicate summary for group "0|com.tencent.android.qqdownloader|g:com.tencent.android.qqdownloader": "0|com.tencent.android.qqdownloader|127|null|10081" vs. "0|com.tencent.android.qqdownloader|115|null|10081" [... cut ...]
> 08-21 03:18:32.914 46385 46853 E AndroidRuntime: at android.net.IVpnManager$Stub.onTransact(IVpnManager.java:286)
> 08-21 03:18:32.914 46385 46853 E AndroidRuntime: at android.os.Binder.execTransactInternal(Binder.java:1179)
> 08-21 03:18:32.914 46385 46853 E AndroidRuntime:
> 08-21 03:18:32.917 237 46855 I DropBoxManagerService: add tag=data_app_crash isTagEnabled=true flags=0x2

modprobe tun

发自我的iPhone

CNflysky, 2024-08-23:
善用issue搜索功能

closed 2024-08-23 by CNflysky