CNflysky/redroid-rk3588 #22: 重返未来1999 游戏闪退

li-chx, 2024-10-19:
**Describe the bug**
**描述Bug**
游戏启动时，过完标题页面，进入资源加载页面后闪退

**Required Information**
**基本信息**
- docker image name
- 镜像名称
`13.0.0-latest`

- Board Information
- 板卡信息
`Orange Pi 5s + 8 GB RAM`

- OS Information
- 系统信息
`Ubuntu22.04-aarch64`

- Kernel version
- 内核版本
`Linux lichx-orangepi 5.10.0-1012-rockchip #12-Ubuntu SMP Wed Aug 14 22:22:22 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
安装并启动 重返未来1999 官服即可

**Screenshots**
**截图**
If applicable, add screenshots to help explain your problem.
无

**Additional context**
**附加说明**
Add any other context about the problem here.
logcat messages are suggested place into pastebin.
log： [pastebin链接](https://pastebin.com/Xath2tiZ)

CNflysky, 2024-10-27:
OK on my device
[image]
check your kernel configuration
#3

li-chx, 2024-10-29:
I guess the wrong value of VA_BITS(48) caused my app to crash. I noticed the problem shortly after submitting the issue, so I closed it. I've been trying to find the right kernel recently, but Armbian build no longer supports 5.10, and I couldn't find the method to modify the kernel options for the ubuntu-rockchip project.  
How you built your system?

CNflysky, 2024-10-29:
> I guess the wrong value of VA_BITS(48) caused my app to crash. I noticed the problem shortly after submitting the issue, so I closed it. I've been trying to find the right kernel recently, but [...]

try this
https://www.nyanners.moe/post/fix-rk3588-redroid-azur-lane-crash.html

li-chx, 2024-10-29:
Solved. Thanks a lot.

closed 2024-10-29 by li-chx