CNflysky/redroid-rk3588 #31: 最新版的QQ打开会显示网络错误不知道是受什么影响了9.1.52

holai, 2025-03-10:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
最新版的QQ打开会显示网络错误不知道是受什么影响了9.1.52
 
**基本信息**
- redroid-rk3588:lineage-20 

- Board Information
- 板卡信息
e.g. `Orange Pi 5 Plus + 16 GB RAM`

- OS Information
- 系统信息
e.g. `Armbian Bookworm Server`

- Kernel version
- 内核版本
e.g. `5.10.160-rockchip-rk35xx`

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
以下是复现该问题的步骤：

最新版QQV9.1.52打开后不能登录

**Screenshots**
**截图**
If applicable, add screenshots to help explain your problem.
如果可以，添加一些截图以更好地解释问题

**Additional context**
**附加说明**
Add any other context about the problem here.
logcat messages are suggested place into pastebin.
如果有额外的问题说明，请放在这里
logcat信息建议放在pastebin里

holai, 2025-03-10:
[image]

CNflysky, 2025-03-11:
enable `FakeWiFi`, then try again.

closed 2025-04-04 by holai