CNflysky/redroid-rk3588 #34: 刷入 Armbian minimal for OPi 5 6.1.84   显示错误

holai, 2025-04-10:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
**描述Bug**
香橙派5Plus 32G  使用md中的Armbian minimal for OPi 5 6.1.84 出现  
ALERT!  UUID=fda9.....     does not exits.  Dropping  to a shell!

**Required Information**
**基本信息**
- docker image name
- 镜像名称 

- Board Information
- 板卡信息
e.g. `Orange Pi 5 Plus + 32 GB RAM`

- OS Information
- 系统信息
e.g. `Armbian Bookworm Server`

- Kernel version
- 内核版本
e.g. `5.10.160-rockchip-rk35xx`

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:

刷入开机进入 initramfs 模式   

**Screenshots**
**截图**
If applicable, add screenshots to help explain your problem.
如果可以，添加一些截图以更好地解释问题

**Additional context**
**附加说明**
Add any other context about the problem here.
logcat messages are suggested place into pastebin.
如果有额外的问题说明，请放在这里
logcat信息建议放在pastebin里

CNflysky, 2025-04-10:
The image is only suitable for `Orange Pi 5`.
try replace `fdtfile` in `/boot/armbianEnv.txt ` to `rockchip/rk3588-orangepi-5-plus.dtb`.

liej6799, 2025-04-17:
Hi @CNflysky, I am able to run your redroid image with `kernel 5.10 ubuntu-rockchip` , I saw you are able to build for `kernel 6.1 armbian` with `dev/dma_heap/system-uncached-dma32 ` that allows mali GPU Acceleration on redroid. 

May I know what are the patches needed to support for `kernel 6.1 armbian`, as currently our board is quite different `orange pi 5 pro`, Can I just change modify the img and change the fdtfile or need to rebuild the image.

CNflysky, 2025-04-17:
> **Edit:**
> 
> I tried rebuild the image with armbian for my board with [this](https://github.com/armbian/build) with latest kernel `6.1.115-vendor-rk35xx`, but the `dev/dma_heap/system-uncached-dma32 ` still missing, can help guide the patch needed to rebuild this image.
> 
> Hi [@CNflysky](https://github.com/CNflysky), I am able to run your redroid image with `kernel 5.10 ubuntu-rockchip` , I saw you are able to build for `kernel 6.1 armbian` with `dev/dma_heap/system-uncached-dma32 ` that allows mali GPU Acceleration on redroid.
> 
> May I know what are the patches needed to support for `kernel 6.1 armbian`, as currently our board is quite different `orange pi 5 pro`, Can I just change modify the img and change the fdtfile or need to rebuild the image.

You need **Customized 6.1 kernel** to run redroid with GPU acceleration. please read REAEME files carefully.

liej6799, 2025-04-17:
Hi @CNflysky, I have fixed the dma issue based on the patch here [here](https://github.com/armbian/linux-rockchip/compare/rk-6.1-rkr4.1...CNflysky:linux-rockchip:rk-6.1-rkr4.1). Able to build with `armbian 6.1.115` for orange pi pro. 

Anyway thanks for your contribution on this project.

closed 2025-04-17 by holai