CNflysky/redroid-rk3588 #37: Mainline this to Armbian for everyone to use??

DarkevilPT, 2025-05-11:
Hi CNflysky.
Before anything I wanna thank you for this amazing development. But I would also love to be able to use this on archlinux and maybe armbian on radxa compute module 5 and im sure others would love this too in other devices.

Is there any remote chance that you can enable this for us all once and for all?

CNflysky, 2025-05-12:
Don't know armbian will accept my patch; it is not fully tested yet

DarkevilPT, 2025-05-12:
I would say give it a try.

[image]

The only feedback I received was from someone more experienced than me with Linux, and they said this. Maybe you’ll know better if any changes are needed.

Still, give it a shot and see what kind of response you get, maybe?

I dont own an orange pi to test it.

CNflysky, 2025-05-12:
> I would say give it a try.
> 
> [image]
> 
> The only feedback I received was from someone more experienced than me with Linux, and they said this. Maybe you’ll know better if any changes are needed.
> 
> Still, give it a shot and see what kind of response you get, maybe?
> 
> I dont own an orange pi to test it.

... the service `android.hardware.graphics.allocator` will allocate graphics(?) memory from `/dev/dma_heap/system-uncached-dma32` device, without that device system will not boot. i've tried change that device to `/dev/dma_heap/cma` or something else, container will crash after started.