CNflysky/redroid-rk3588 #40: 绿联DH 4300 Plus 考虑增加兼容吗？

coracoo, 2025-05-30:
大佬好，前端时间发布的绿联DH4300 Plus，采用了RK3588C处理器，我尝试了firefly AIC 和 redroid项目，发现都无法成功，原因可能在于其产品的内核编译问题。

对于成品设备内核问题，我估计是应该没办法解决，而且项目上也提供了建议使用的系统版本和内核。

所以，探讨性的咨询一下，是否有其它方式可以解决？因为该产品用户还是比较多，如果可以类似极空间T2S那种有云安卓手机的话，可玩性可以大大增加。

1.使用检测脚本展示如下：

```
root@DH4300PLUS-B1AB:/volume1/docker/redroid# ./envcheck.sh 
========================================
redroid-rk3588 environment check script
========================================
checking kernel config location...
Kernel config: /proc/config.gz
========================================
checking mali gpu driver...
CONFIG_MALI_BIFROST=y
========================================
checking mali firmware...
Mali CSF Firmware exists.
========================================
checking kernel features...
CONFIG_ANDROID_BINDERFS is not enabled in your kernel.
CONFIG_PSI=y
CONFIG_ARM64_VA_BITS=39
========================================
checking binderfs...
binderfs not enabled.
========================================
checking required dma-heap devices...
dma-buf device is not exist.
========================================
Summary
FATAL: required dma-buf heap device missing
FATAL: CONFIG_ANDROID_BINDERFS is not enabled in your kernel
FATAL: At least one of those mandatory kernel features are not met. You must find another kernel built with those features or build a customized kernel by yourself.
```

2.dma-buf 经过我测试，是可以开启。
```
zcat /proc/config.gz | grep CONFIG_DMABUF_HEAPS
CONFIG_DMABUF_HEAPS=y
CONFIG_DMABUF_HEAPS_SYSTEM=y
CONFIG_DMABUF_HEAPS_CMA=y
# CONFIG_DMABUF_HEAPS_SRAM is not set
# CONFIG_DMABUF_HEAPS_ROCKCHIP is not set
```

3.binderfs因为内核问题，应该没办法处理。

4.容器我部署成功了，并且端口也映射了出去，但是无法使用adb链接，感觉是容器虽然跑起来了，但是实际成功运行导致的？
```
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:6379          0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:33039           0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:9999            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:9443            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5555            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5443            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5443            0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      -                   
tcp6       0      0 :::22                   :::*                    LISTEN      -                   
tcp6       0      0 :::111                  :::*                    LISTEN      -                   
tcp6       0      0 :::80                   :::*                    LISTEN      -                   
tcp6       0      0 :::443                  :::*                    LISTEN      -                   
tcp6       0      0 :::9999                 :::*                    LISTEN      -                   
tcp6       0      0 :::50349                :::*                    LISTEN      -                   
tcp6       0      0 :::9443                 :::*                    LISTEN      -                   
tcp6       0      0 ::1:6379                :::*                    LISTEN      -                   
tcp6       0      0 :::5555                 :::*                    LISTEN      -                   
tcp6       0      0 :::5443                 :::*                    LISTEN      -                   
tcp6       0      0 :::5443                 :::*                    LISTEN      -    
```

CNflysky, 2025-05-30:
没有。缺少必须的内核特性的话，只能通过重新编译内核来解决。

coracoo, 2025-05-30:
感谢感谢。那只能沟通看看绿联那边后面能否添加binderfs了。现在看无法连接的原因是因为容器一直在重启，应该就是缺少模块导致的

CNflysky, 2025-05-30:
> 2.dma-buf 经过我测试，是可以开启。
> 
> ```
> zcat /proc/config.gz | grep CONFIG_DMABUF_HEAPS
> CONFIG_DMABUF_HEAPS=y
> CONFIG_DMABUF_HEAPS_SYSTEM=y
> CONFIG_DMABUF_HEAPS_CMA=y
> # CONFIG_DMABUF_HEAPS_SRAM is not set
> # CONFIG_DMABUF_HEAPS_ROCKCHIP is not set
> ```
这里默认的dmabuf少东西 必须要把rk搞的那个dmabuf搞来

coracoo, 2025-05-30:
明白了，那还是需要官方下场来弄。自己不敢编译内核，怕把系统搞崩了，谢谢大佬，关闭issue了~

closed 2025-05-30 by coracoo