CNflysky/redroid-rk3588 #41: Magisk not work

yigger, 2025-06-20:
**NOTICE: ANY BUG REPORT NOT USING THIS TEMPLETE WILL BE CLOSED AS NOT PLANNED**
**注意：任何未使用该模板的错误报告将被直接关闭**

**Describe the bug**
**描述Bug**
手上刚好有一块NanoPC-T6（https://[wiki.friendlyelec.com/wiki/index.php/NanoPC-T6/zh](https://wiki.friendlyelec.com/wiki/index.php/NanoPC-T6/zh)），想着测试一下，但是容器运行后，Magisk 无论怎么折腾都是没有 root 的状态，导致无法安装 LSP 模块

**Required Information**
**基本信息**
- docker image name
- 镜像名称
cnflysky/redroid-rk3588:lineage-20 

- Board Information
- 板卡信息
`NanoPi T6 + 16 GB RAM`

- OS Information
- 系统信息

- Kernel version
- 内核版本
6.1.99-rockchip-rk3588

**To Reproduce**
**复现步骤**
Steps to reproduce the behavior:
以下是复现该问题的步骤：

```
docker run -d --privileged -p 5588:5555 -v ./data:/data -v /dev/mali0:/dev/mali0 --name redroid docker.xuanyuan.me/cnflysky/redroid-rk3588:lineage-20 androidboot.redroid_height=1920 androidboot.redroid_width=1080 androidboot.redroid_magisk=1 androidboot.redroid_virt_wifi=1 androidboot.redroid_gpu_mode=mali
```

Magisk Log:
```

---Manager Logs---
f2357dbc (27001)

--------- beginning of main
06-20 00:58:44.056  3962  3962 E .huskydg.magisk: Not starting debugger since process cannot load the jdwp agent.
06-20 00:58:44.077  3962  3962 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10126; state: ENABLED
06-20 00:58:44.098  3962  3962 V GraphicsEnvironment: ANGLE Developer option for 'io.github.huskydg.magisk' set to: 'default'
06-20 00:58:44.099  3962  3962 V GraphicsEnvironment: ANGLE GameManagerService for io.github.huskydg.magisk: false
06-20 00:58:44.099  3962  3962 V GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
06-20 00:58:44.103  3962  3962 D NetworkSecurityConfig: No Network Security Config specified, using platform default
06-20 00:58:44.103  3962  3962 D NetworkSecurityConfig: No Network Security Config specified, using platform default
06-20 00:58:44.139  3962  3981 D ProfileInstaller: Skipping profile installation for io.github.huskydg.magisk
06-20 00:58:44.144  3962  3962 D AppCompatDelegate: Checking for metadata for AppLocalesMetadataHolderService : Service not found
06-20 00:58:44.151  3962  3962 I AppCompatDelegate: The Activity's LayoutInflater already has a Factory installed so we can not install AppCompat's
06-20 00:58:44.166  3962  3962 D CompatibilityChangeReporter: Compat change id reported: 237531167; UID 10126; state: DISABLED
06-20 00:58:44.173  3962  3988 W Parcel  : Expecting binder but got null!
06-20 00:58:44.186  3962  3988 I mali_so : [File] : hardware/arm/maliT760/driver/product/base/src/mali_base_kbase.c; [Line] : 1392; [Func] : base_context_deal_with_version_affairs_rk_ext;
06-20 00:58:44.186  3962  3988 I mali_so : arm_release_ver of this mali_so is 'g18p0-01eac0', rk_so_ver is '2 '.
06-20 00:58:44.186  3962  3988 I mali_so : [File] : hardware/arm/maliT760/driver/product/base/src/mali_base_kbase.c; [Line] : 1408; [Func] : base_context_deal_with_version_affairs_rk_ext;
06-20 00:58:44.186  3962  3988 I mali_so : arm_release_vers are match. to set the full mali_ver 'g18p0-01eac0-2' as value of 'sys.gmali.version'.
06-20 00:58:44.206  3962  3988 E OpenGLRenderer: Unable to match the desired swap behavior.
06-20 00:58:44.264  3962  3980 D CompatibilityChangeReporter: Compat change id reported: 194532703; UID 10126; state: ENABLED
06-20 00:58:44.290  3962  3980 D nativeloader: classloader namespace configured for unbundled product apk. library_path=/system/product/priv-app/GmsCore/lib/arm64:/system/product/priv-app/GmsCore/GmsCore.apk!/lib/arm64-v8a:/product/lib64:/system/product/lib64
06-20 00:58:44.296  3962  3980 V NativeCrypto: Registering com/google/android/gms/org/conscrypt/NativeCrypto's 295 native methods...
06-20 00:58:44.309  3962  3980 I ProviderInstaller: Installed default security provider GmsCore_OpenSSL
06-20 00:58:44.321  3962  3962 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
06-20 00:58:44.321  3962  3962 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
06-20 00:58:44.332  3962  3962 D CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10126; state: ENABLED
06-20 00:58:44.337  3962  3962 D CompatibilityChangeReporter: Compat change id reported: 210923482; UID 10126; state: ENABLED
06-20 00:58:44.459  3962  4015 D TrafficStats: tagSocket(91) with statsTag=0xffffffff, statsUid=-1
06-20 00:58:45.209  3962  3962 E Ix      : a.ks: HTTP 404 
06-20 00:58:45.209  3962  3962 E Ix      : 	at a.uo.z(SourceFile:143)
06-20 00:58:45.209  3962  3962 E Ix      : 	at a.iq.r(SourceFile:9)
06-20 00:58:45.209  3962  3962 E Ix      : 	at a.Cx.run(SourceFile:57)
06-20 00:58:45.209  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
06-20 00:58:45.209  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
06-20 00:58:45.209  3962  3962 E Ix      : 	at java.lang.Thread.run(Thread.java:1012)
06-20 00:58:45.823  3962  3988 W Parcel  : Expecting binder but got null!
06-20 00:58:46.763  3962  3962 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
06-20 00:58:46.763  3962  3962 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
06-20 00:58:47.612  3962  3962 E Ix      : a.ks: HTTP 404 
06-20 00:58:47.612  3962  3962 E Ix      : 	at a.uo.z(SourceFile:143)
06-20 00:58:47.612  3962  3962 E Ix      : 	at a.iq.r(SourceFile:9)
06-20 00:58:47.612  3962  3962 E Ix      : 	at a.Cx.run(SourceFile:57)
06-20 00:58:47.612  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
06-20 00:58:47.612  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
06-20 00:58:47.612  3962  3962 E Ix      : 	at java.lang.Thread.run(Thread.java:1012)
06-20 00:58:47.661  3962  3962 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
06-20 00:58:47.661  3962  3962 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
06-20 00:58:49.059  3962  3962 E Ix      : a.ks: HTTP 404 
06-20 00:58:49.059  3962  3962 E Ix      : 	at a.uo.z(SourceFile:143)
06-20 00:58:49.059  3962  3962 E Ix      : 	at a.iq.r(SourceFile:9)
06-20 00:58:49.059  3962  3962 E Ix      : 	at a.Cx.run(SourceFile:57)
06-20 00:58:49.059  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
06-20 00:58:49.059  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
06-20 00:58:49.059  3962  3962 E Ix      : 	at java.lang.Thread.run(Thread.java:1012)
06-20 00:58:50.090  3962  3988 W Parcel  : Expecting binder but got null!
06-20 00:58:50.106  3962  3988 E OpenGLRenderer: Unable to match the desired swap behavior.
06-20 00:59:14.609  3962  3971 W .huskydg.magisk: Cleared Reference was only reachable from finalizer (only reported once)
06-20 01:04:19.115  3962  4664 D TrafficStats: tagSocket(5) with statsTag=0xffffffff, statsUid=-1
06-20 01:04:19.142  3962  3988 E OpenGLRenderer: Unable to match the desired swap behavior.
06-20 01:04:19.836  3962  3962 E Ix      : a.ks: HTTP 404 
06-20 01:04:19.836  3962  3962 E Ix      : 	at a.uo.z(SourceFile:143)
06-20 01:04:19.836  3962  3962 E Ix      : 	at a.iq.r(SourceFile:9)
06-20 01:04:19.836  3962  3962 E Ix      : 	at a.Cx.run(SourceFile:57)
06-20 01:04:19.836  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
06-20 01:04:19.836  3962  3962 E Ix      : 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
06-20 01:04:19.836  3962  3962 E Ix      : 	at java.lang.Thread.run(Thread.java:1012)
06-20 01:06:21.847  3962  3962 W OnBackInvokedCallback: OnBackInvokedCallback is not enabled for the application.
06-20 01:06:21.847  3962  3962 W OnBackInvokedCallback: Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
```

**Screenshots**
**截图**

[image]

[image]

[image]

yigger, 2025-06-20:
Sorry..  here is the logs.

```
pi@NanoPC-T6-LTS:~/redroid/test_d$ docker exec redroid logcat | grep magisk | tee -a log.txt
06-20 11:03:11.635  3724  3724 I Finsky  : [2] yfh.c(8): VerifyApps:, Install-time verification requested for package io.github.huskydg.magisk, id = 0
06-20 11:03:21.656  3724  3724 I Finsky  : [2] VerifyInstallTask.adS(7): VerifyApps: Verification complete: id=0, package_name=io.github.huskydg.magisk
06-20 11:03:21.832   249   297 W ziparchive: Unable to open '/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.dm': No such file or directory
06-20 11:03:21.833   249   297 I PackageDexOptimizer: Running dexopt (dexoptNeeded=1) on: /data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.apk pkg=io.github.huskydg.magisk isa=arm dexoptFlags=boot_complete,profile_guided,public targetFilter=speed-profile oatDir=/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/oat classLoaderContext=PCL[]{}
06-20 11:03:21.834   176   181 V installd: DexInv: --- BEGIN '/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.apk' ---
06-20 11:03:21.834   176   181 V installd: Running /apex/com.android.art/bin/dex2oat32 in=base.apk out=/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/oat/arm/base.odex
06-20 11:03:21.904  4840  4840 I dex2oat32: /apex/com.android.art/bin/dex2oat32 --input-vdex-fd=-1 --output-vdex-fd=9 --classpath-dir=/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig== --class-loader-context=PCL[]{} --compact-dex-level=none --compiler-filter=speed-profile --compilation-reason=install --max-image-block-size=524288 --resolve-startup-const-strings=true --generate-mini-debug-info
06-20 11:03:22.235   176   181 V installd: DexInv: --- END '/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.apk' (success) ---
06-20 11:03:22.258   249   297 W ziparchive: Unable to open '/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.dm': No such file or directory
06-20 11:03:22.258   249   297 I PackageDexOptimizer: Running dexopt (dexoptNeeded=1) on: /data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.apk pkg=io.github.huskydg.magisk isa=arm64 dexoptFlags=boot_complete,profile_guided,public targetFilter=speed-profile oatDir=/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/oat classLoaderContext=PCL[]{}
06-20 11:03:22.259   176   181 V installd: DexInv: --- BEGIN '/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.apk' ---
06-20 11:03:22.259   176   181 V installd: Running /apex/com.android.art/bin/dex2oat32 in=base.apk out=/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/oat/arm64/base.odex
06-20 11:03:22.299  4859  4859 I dex2oat32: /apex/com.android.art/bin/dex2oat32 --input-vdex-fd=-1 --output-vdex-fd=9 --classpath-dir=/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig== --class-loader-context=PCL[]{} --compact-dex-level=none --compiler-filter=speed-profile --compilation-reason=install --max-image-block-size=524288 --resolve-startup-const-strings=true --generate-mini-debug-info
06-20 11:03:22.567   176   181 V installd: DexInv: --- END '/data/app/~~OJRusseqm4N5sRT5afQT5w==/io.github.huskydg.magisk-Mo-G9QXpcSui62oKEDCxig==/base.apk' (success) ---
06-20 11:03:22.592   249   297 V BackupManagerService: [UserID:0] restoreAtInstall pkg=io.github.huskydg.magisk token=1 restoreSet=0
06-20 11:03:22.604  3333  3333 I MediaProvider: Invalidating LocalCallingIdentity cache for package io.github.huskydg.magisk. Reason: package android.intent.action.PACKAGE_ADDED
06-20 11:03:22.614  3724  3724 I Finsky  : [2] pft.n(8): Invalidating cached PackageState for io.github.huskydg.magisk
06-20 11:03:22.639  3237  4773 I FontLog : Package io.github.huskydg.magisk has no metadata [CONTEXT service_id=132 ]
06-20 11:03:22.663   249  3317 W BackupTransportManager: Package io.github.huskydg.magisk enabled setting: 0
06-20 11:03:22.668  3724  3724 I Finsky  : [2] pft.n(8): Invalidating cached PackageState for io.github.huskydg.magisk
06-20 11:03:22.723  3724  4893 I Finsky  : [138] ezp.accept(28): Asset module storage cleared for package io.github.huskydg.magisk.
06-20 11:03:35.679   249   748 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=io.github.huskydg.magisk/com.topjohnwu.magisk.ui.MainActivity bnds=[226,434][435,674]} from uid 10104
06-20 11:03:35.724   249   617 D CoreBackPreview: Window{4beb653 u0 Splash Screen io.github.huskydg.magisk}: Setting back callback OnBackInvokedCallbackInfo{mCallback=android.window.IOnBackInvokedCallback$Stub$Proxy@648cc89, mPriority=0}
06-20 11:03:35.742   249   277 I ActivityManager: Start proc 4973:io.github.huskydg.magisk/u0a126 for next-top-activity {io.github.huskydg.magisk/com.topjohnwu.magisk.ui.MainActivity}
06-20 11:03:35.782  4973  4973 E .huskydg.magisk: Not starting debugger since process cannot load the jdwp agent.
06-20 11:03:35.822  4973  4973 V GraphicsEnvironment: ANGLE Developer option for 'io.github.huskydg.magisk' set to: 'default'
[... cut ...] 09c468a50
06-20 11:03:37.859   249   361 D InputDispatcher:     * [TOUCHED] package=io.github.huskydg.magisk/10126, id=656, mode=BLOCK_UNTRUSTED, alpha=1.00, frame=[0,0][1080,1920], touchableRegion=[0,0][1080,1920], window={9c62616 io.github.huskydg.magisk/com.topjohnwu.magisk.ui.MainActivity}, inputConfig={0x0}, hasToken=true, applicationInfo.name=ActivityRecord{fd046db u0 io.github.huskydg.magisk/com.topjohnwu.magisk.ui.MainActivity} t126}, applicationInfo.token=0xb40000709c468a50
06-20 11
```

CNflysky, 2025-06-20:
i don't see any clear error; try capture again with `grep -i magisk`

CNflysky, 2025-06-21:
`adb shell getprop ro.boot.redroid_magisk` ？if this is true/1 then magiskd is working abnormaly.

CNflysky, 2025-06-21:
check issue #4

yigger, 2025-06-21:
emm.. I also saw this issue a few days ago, but I saw that the kernel mentioned was 6.1. Mine is 6.1.99. I don't think this is the problem. I will try to downgrade the kernel（the firmware downgrade is a litter difficult）. Thank you for your reply and help.

CNflysky, 2025-06-21:
can't continue trouble shooting without log. take a look at `/vendor/etc/init/init.magisk.rc`

yigger, 2025-06-21:
ok, thanks. I'll try to check it out, and I found a similar problem [issue_linik](https://github.com/ayasa520/redroid-script/issues/28), but the redroid version is different, and the actual situation may be different.

But there is a question, if i need to change the file, can i replace it by editing the dockerfile like that repository script?  it will be work at the image?

```dockerfile
from cnflysky/redroid-rk3588:lineage-20
COPY magisk /xxx
```

CNflysky, 2025-06-21:
you can export image out, unpack and do the patch, then pack it back to image and import into system.

closed 2026-03-17 by yigger